atenpas/gpd #107: Compiling failed

gachiemchiep, 2021-01-08:
Hello
With the newest commit, I have the following error when compiling 

```bash
jil@tr_ws:~/opt/gpd/gpd.build$ 
jil@tr_ws:~/opt/gpd/gpd.build$ make
[  3%] Built target gpd_eigen_utils
[  6%] Built target gpd_point_list
[  9%] Built target gpd_config_file
[ 13%] Built target gpd_image_geometry
[ 16%] Built target gpd_antipodal
[ 19%] Built target gpd_finger_hand
[ 22%] Built target gpd_hand
[ 26%] Built target gpd_hand_geometry
[ 29%] Built target gpd_local_frame
[ 32%] Built target gpd_hand_set
[ 36%] Built target gpd_cloud
[ 39%] Built target gpd_frame_estimator
[ 42%] Built target gpd_plot
[ 45%] Built target gpd_hand_search
[ 49%] Built target gpd_candidates_generator
Scanning dependencies of target gpd_generate_candidates
[ 50%] Building CXX object CMakeFiles/gpd_generate_candidates.dir/src/generate_candidates.cpp.o
[ 52%] Linking CXX executable generate_candidates
/tmp/ccB5MS1V.ltrans1.ltrans.o: In function `gpd::apps::generate_candidates::DoMain(int, char**)':
/home/jil/opt/gpd/gpd/src/generate_candidates.cpp:48: undefined reference to `gpd::util::ConfigFile::ConfigFile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/home/jil/opt/gpd/gpd/src/generate_candidates.cpp:49: undefined reference to `gpd::util::ConfigFile::ExtractKeys()'
/home/jil/opt/gpd/gpd/src/generate_candidates.cpp:51: undefined reference to `gpd::candidate::HandGeometry::HandGeometry()'
/home/jil/opt/gpd/gpd/src/generate_candidates.cpp:75: undefined reference to `gpd::util::ConfigFile::getValueOfKeyAsString(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/home/jil/opt/gpd/gpd/src/generate_candidates.cpp:77: undefined reference to `gpd::util::ConfigFile::getValueOfKeyAsString(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/home/jil/opt/gpd/gpd/src/generate_candidates.cpp:93: undefined reference to `gpd::util::ConfigFile::getValueOfKeyAsStdVectorInt(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/tmp/ccB5MS1V.ltrans1.ltrans.o: In function `__base_ctor ':
/home/jil/opt/gpd/gpd/include/gpd/candidate/hand_search.h:76: undefined reference to `gpd::candidate::HandGeometry::HandGeometry()'
/tmp/ccB5MS1V.ltrans1.ltrans.o: In function `gpd::apps::generate_candidates::DoMain(int, char**)':
/home/jil/opt/gpd/gpd/src/generate_candidates.cpp:128: undefined reference to `gpd::candidate::CandidatesGenerator::CandidatesGenerator(gpd::candidate::CandidatesGenerator::Parameters const&, gpd::candidate::HandSearch::Parameters const&)'
/home/jil/opt/gpd/gpd/src/generate_candidates.cpp:135: undefined reference to `gpd::util::Cloud::Cloud(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Eigen::Matrix<double, 3, -1, 0, 3, -1> const&)'
/home/jil/opt/gpd/gpd/src/generate_candidates.cpp:143: undefined reference to `gpd::util::Cloud::setNormalsFromFile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/home/jil/opt/gpd/gpd/src/generate_candidates.cpp:149: undefined reference to `gpd::candidate::CandidatesGenerator::preprocessPointCloud(gpd::util::Cloud&)'
/home/jil/opt/gpd/gpd/src/generate_candidates.cpp:153: undefined reference to `gpd::candidate::CandidatesGenerator::generateGraspCandidates(gpd::util::Cloud const&)'
/home/jil/opt/gpd/gpd/src/generate_candidates.cpp:158: undefined reference to `gpd::util::Plot::plotFingers3D(std::vector<std::unique_ptr<gpd::candidate::Hand, std::default_delete<gpd::candidate::Hand> >, std::allocator<std::unique_ptr<gpd::candidate::Hand, std::default_delete<gpd::candidate::Hand> > > > const&, boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGBA> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, gpd::candidate::HandGeometry const&, bool)'
/tmp/ccB5MS1V.ltrans5.ltrans.o: In function `double gpd::util::ConfigFile::getValueOfKey<double>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, double const&) const [clone .constprop.20]':
/home/jil/opt/gpd/gpd/include/gpd/util/config_file.h:84: undefined reference to `gpd::util::ConfigFile::keyExists(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
/tmp/ccB5MS1V.ltrans5.ltrans.o: In function `int gpd::util::ConfigFile::getValueOfKey<int>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int const&) const [clone .constprop.17]':
/home/jil/opt/gpd/gpd/include/gpd/util/config_file.h:84: undefined reference to `gpd::util::ConfigFile::keyExists(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
/tmp/ccB5MS1V.ltrans5.ltrans.o: In function `bool gpd::util::ConfigFile::getValueOfKey<bool>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool const&) const [clone .constprop.14]':
/home/jil/opt/gpd/gpd/include/gpd/util/config_file.h:84: undefined reference to `gpd::util::ConfigFile::keyExists(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
collect2: error: ld returned 1 exit status
CMakeFiles/gpd_generate_candidates.dir/build.make:299: recipe for target 'generate_candidates' failed
make[2]: *** [generate_candidates] Error 1
CMakeFiles/Makefile2:147: recipe for target 'CMakeFiles/gpd_generate_candidates.dir/all' failed
make[1]: *** [CMakeFiles/gpd_generate_candidates.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

atenpas, 2021-01-10:
What's your gcc version and operating system?

Have you tried "cleaning up" the make and cmake commands? Like running `make clean` and/or deleting the build folder.

sisco0, 2021-01-11:
It pretty seems that linker is not correctly finding the way to compile the project. As stated by @atenpas a clean build should be the way to go, removing the build folder and building anewly.
Seems to compile correctly under Ubuntu 20.04 (WSL 2).

`cmake --version` is 3.16.3
`g++ --version` is 9.3.0

gachiemchiep, 2021-01-14:
@atenpas @sisco0 
I'm using Ubuntu-16.04 with Cmake 3.16.2
I tried to clean the build directory then rebuild with gcc-6 and gcc-7 but the same errors happened.. 

[gcc-7.log](https://github.com/atenpas/gpd/files/5813801/gcc-7.log)
[gcc-6.log](https://github.com/atenpas/gpd/files/5813802/gcc-6.log)

closed 2021-01-14 by gachiemchiep