atenpas/gpd #108: How can I modified the cfg file to get correct result from table_mug.pcd

mahaoxiang822, 2021-01-12:
Same as #89, I get many strange grasps far from the plane. How to change the eigen_params.cfg to get correct result for point clouds like table_mug.pcd?

atenpas, 2021-01-22:
[My reply](https://github.com/atenpas/gpd/issues/89#issuecomment-603923749) in https://github.com/atenpas/gpd/issues/89 describes one approach for this. I would suggest that you read and play around with the parameters in the config file. Basically, you want to tell the detector where to sample (on the mug). One way to do this is to figure out where the table plane is.

Siddhartha-techolution, 2024-03-13:
can i know how to figure this thing
i was trying from a lot of days but unable to configure it