atenpas/gpd #111: What is the format of the dataset file in step 9?

kyrie-H, 2021-01-23:
I have downloaded the bigbird dataset,when I execute "./generate_data ../cfg/generate_data.cfg"
#grasps :0.#positives:0,#negatives:0
An error occurred：Point cloud is empty! Eval GT ...

I want to know how to configure dataset files.
Hope to get your advice, thank you very much！

qyp-robot, 2023-03-21:
like this :
[image]

Gerjen100, 2023-12-11:
Hi! i was also testing this code, i downloaded the dataset and out it in a folder. Then i named the objects in the objects.txt file. But i got the same error, did you manage to resolve this? Also there don't seem to be any meshes for the ground truth in the BigBird dataset. Did you download these from somewhere?