atenpas/gpd #114: The error occurs while building the projects of the gpg

yangyufeng-6666, 2021-02-21:
yangyufeng@yangyufeng-Inspiron-5577:~/gpg/build$ make
[  7%] Built target gpg_local_frame
[ 14%] Built target gpg_eigen_utils
[ 21%] Built target gpg_point_list
[ 28%] Built target gpg_antipodal
[ 35%] Built target gpg_finger_hand
[ 42%] Built target gpg_grasp
[ 50%] Built target gpg_grasp_set
[ 57%] Built target gpg_cloud_camera
Scanning dependencies of target gpg_plot
[ 60%] Building CXX object CMakeFiles/gpg_plot.dir/src/gpg/plot.cpp.o
/home/yangyufeng/gpg/src/gpg/plot.cpp: In member function ‘void Plot::runViewer(boost::shared_ptr<pcl::visualization::PCLVisualizer>&) const’:
/home/yangyufeng/gpg/src/gpg/plot.cpp:454:12: error: ‘boost::this_thread’ has not been declared
  454 |     boost::this_thread::sleep(boost::posix_time::microseconds(100000));
      |            ^~~~~~~~~~~
make[2]: *** [CMakeFiles/gpg_plot.dir/build.make:63：CMakeFiles/gpg_plot.dir/src/gpg/plot.cpp.o] 错误 1
make[1]: *** [CMakeFiles/Makefile2:311：CMakeFiles/gpg_plot.dir/all] 错误 2
make: *** [Makefile:130：all] 错误 2

Does anyone meets the same problem?

yxk-tian, 2021-05-17:
I have the same problem, and i soluved that by reinstall the system. My system is ubuntu 18.04

hqf2591102423, 2022-01-24:
> yangyufeng@yangyufeng-Inspiron-5577:~/gpg/build$ make [ 7%] Built target gpg_local_frame [ 14%] Built target gpg_eigen_utils [ 21%] Built target gpg_point_list [ 28%] Built target gpg_antipodal [ [...]

have you solve the problem