atenpas/gpd #116: datasets generator   program

wuxiangrui188, 2021-04-21:
First,thank you for your amazing work!
could you send the program for generate  datasetes(like the bigbird datasets) ? Or samething like the datasets file construct?
In my opinion ,the datasets structure should be include:(example 1 object)
0_gt.pcd   0_gt.csv
0_1.pcd 0_2.pcd ......0_20.pcd (its corresponding .csv file for store normal data)

it is that true?

hope for your helpping!

zhoumiaoqiang, 2021-12-28:
> First,thank you for your amazing work! could you send the program for generate datasetes(like the bigbird datasets) ? Or samething like the datasets file construct? In my opinion ,the datasets [...]

Hello, I am also confused about this. Have you completed the generation of the data set?

wuxiangrui188, 2022-01-03:
> Hello, I am also confused about this. Have you completed the generation of the data set?

yes,i just use the function in the program of other files to generate the Normal vectors. you can try again!

zhoumiaoqiang, 2022-01-04:
> yes,i just use the function in the program of other files to generate the Normal vectors. you can try again!

Can you add a contact number?QQ783145004

huiwenzhang, 2023-02-07:
> 

@TechyRio  The data format is exactly as you pasted above according to the code. However, the bigbird dataset format is very different from the format used in this repo. Which program did you use to generate the dataset? Could U explain it more clearly?

qyp-robot, 2023-03-21:
@huiwenzhang @zhoumiaoqiang 
hi！ ，I find the object data from bigbird contain conpleted mesh module and partial point cloud, and the generate c++ code need 20 numbers point cloud module, when I use the parcial point cloud that view by singel camera, the antipodal graspa does not exist；I know need two singel camera point cloud connect ; or I can ues complete data to train? like this:
[image]

qyp-robot, 2023-03-21:
can you help me ?thank you !

huiwenzhang, 2023-04-19:
@qyp-robot   which c++ file you used for generating data? Basically, U don't need two cameras. The full mesh model is used to evaluate valid antipodal grasps, i.e. the label for sampled grasps. U can't train with full mesh model, except that you can get the full mesh when doing inference in real world.