atenpas/gpd #123: How can I add additional filters/filtering criterion based on their position on object ?

Gaurav37, 2022-01-02:
Hi,
I would like to know how can I edit some grasps out ? Is there a position where I can store the grasps generated and then cut out those grasps which I don't need?

ojijinWang, 2022-04-07:
I am not @atenpas. lol. you have several options to remove them. one, use the parameter of workspace to filter out the grasp candidates out of the workspace. second, using the parameter of camera positions to filter out the wrong direction. third. i just mention it here because we have not uploaded the code yet. use the paper of our following up research "Robust grasp detection with incomplete point cloud and complex background". If you want i will discuss with the co-authour about sending you the code that we have not published yet. I replay this because we found the same problem and the filter of workspace is not good enough. And we proposed the occlusion checking athough the reviewers said it might not realy necessary.

closed 2022-10-06 by atenpas

Gaurav37, 2022-10-06:
@atenpas  Sorry I forgot to reply and close issue. Yes I implemented it for my use case.
Thanks

lwcode, 2023-11-09:
@ojijinWang  Could I get your code of  "Robust grasp detection with incomplete point cloud and complex background" for reference?

ojijinWang, 2023-11-09:
> @ojijinWang Could I get your code of "Robust grasp detection with incomplete point cloud and complex background" for reference?
@lwcode
Okay, please wait for a week, I am busy. I've not organized the code. Please notify me if I forget.

lwcode, 2023-11-09:
> > @ojijinWang Could I get your code of "Robust grasp detection with incomplete point cloud and complex background" for reference?
> > @lwcode
> > Okay, please wait for a week, I am busy. I've not organized the code. Please notify me if I forget.

@ojijinWang Thank you so much!

ojijinWang, 2023-11-21:
@lwcode
I quickly made a repository for you. It might be buggy. Please try it first.
https://github.com/ojijinWang/gpd_vcloud

I will build a standalone filter that can be applied to any grasp detection method later. Might take a week.