atenpas/gpd #125: In the code, the discrimination about whether the grabbed object is closed by force or not

qyp-robot, 2022-03-17:
Hello!@atenpas 
I'm studying your article: grab pose detection in point clouds; However, I am looking for the judging conditions for judging whether the grabbing candidate object is force-closed grabbing, which is in your code; I see this antipodal.cpp file, and I'm not sure whether this is the judgment of force closure. What is the difference between half grasp and full grasp? What is the code file about force closure judgment? I would appreciate it if you could help me!

qyp-robot, 2022-03-17:
> I'm in a hurry. Please give me some guidance when you have time. Thank you very much!

qyp-robot, 2022-03-17:
[image]

atenpas, 2022-10-06:
I would suggest to have a look at our [conference paper](https://arxiv.org/abs/1706.09911) for this and the referenced literature in the corresponding sections. A _half grasp_ is a grasp for which **only one** finger of the two-finger hand satisfies the conditions for an antipodal grasp.