atenpas/gpd #127: Calculation of principal curvature direction

qyp-robot, 2022-07-13:
hi,To find the normal direction of a surface at a certain point is to fit a plane, the corresponding minimum eigenvalue is the normal vector, then the fork product of the eigenvectors corresponding to the maximum and minimum eigenvalues is the direction of the principal curvature axis？
[image]

According to my understanding, the two real eigenvalues are computed through the Weingarten transform in order to obtain the two principal curvature axis directions

qyp-robot, 2022-07-13:
@atenpas I look forward to your help, thank you very Much!

atenpas, 2022-10-17:
What's a fork product? I think you mean the [cross product](https://en.wikipedia.org/wiki/Cross_product). 

This is just one way of calculating the curvature directions of a surface.