atenpas/gpd #128: Where I can find the output file when using "detect_grasps" ?

tycoer, 2022-07-15:
I'd like to generate grasps for a single object by "detect_grasps", but I cannot find any output file.
How I can generate grasps then get the output file of generated grasps like https://tams.informatik.uni-hamburg.de/research/datasets/PointNetGPD_grasps_dataset.zip for a single object. May be a h5 file or something

WeisserHase-zhang, 2023-11-20:
Hello,

i tried to generate the list of selected grasps, but i just find the img of grasps. how can we output the list of grasps?maybe position and orientation

rojomeng, 2024-04-09:
Hi!

I'm stuck in the same situation, i want to obtain the position and orientation of the gripper but there's no output file like a point cloud that can be used. 
Were you guys able to solve it?

WeisserHase-zhang, 2024-04-09:
> Hi!
> 
> I'm stuck in the same situation, i want to obtain the position and orientation of the gripper but there's no output file like a point cloud that can be used. Were you guys able to solve it?

I called the libgpd_c.so in the script and output the pose.

rojomeng, 2024-04-09:
@WeisserHase-zhang is that a file provided in the same package? I can't find it.

huanzhang-rwth, 2024-04-10:
@rojomeng yes, you can find it after building the package.

ximiluuuu, 2024-12-19:
> @rojomeng yes, you can find it after building the package.是的，您可以在构建软件包后找到它。

I find libgpd_c.so，but how to use it？just in `/gpd/build` print `./libgpd_c.so` ?