atenpas/gpd #131: [pcl::NormalEstimationOMP::compute] input_ is empty!

xiaofeiso, 2022-11-23:
When I try my own point cloud，I run ./detect_grasps ../cfg/eigen_params.cfg,/home/xiaofeisong/Desktop/f/code/our_dataset/luosuan/google_512k/nontextured.pcd

Failed to find match for field 'rgba'.
Loaded point cloud with 76290 points 
============ HAND GEOMETRY ======================
finger_width: 0.01
hand_outer_diameter: 0.12
hand_depth: 0.06
hand_height: 0.02
init_bite: 0.01
=================================================
============ PLOTTING ========================
plot_normals: true
plot_samples false
plot_candidates: true
plot_filtered_candidates: false
plot_valid_grasps: false
plot_clustered_grasps: false
plot_selected_grasps: true
==============================================
============ CLOUD PREPROCESSING =============
voxelize: true
voxel_size: 0.003
remove_outliers: false
workspace: -1.00 1.00 -1.00 1.00 -1.00 1.00 
sample_above_plane: false
normals_radius: 0.030
refine_normals_k: 0
==============================================
============ CANDIDATE GENERATION ============
num_samples: 30
num_threads: 4
nn_radius: 0.01
hand axes: 2 
num_orientations: 8
num_finger_placements: 10
deepen_hand: true
friction_coeff: 20.00
min_viable: 6
==============================================
============ GRASP IMAGE GEOMETRY ===============
volume width: 0.1
volume depth: 0.06
volume height: 0.02
image_size: 60
image_num_channels: 15
=================================================
NET SETUP runtime: 0.0433275
============ CLASSIFIER ======================
model_file: 
weights_file: ../models/lenet/15channels/params/
batch_size: 1
==============================================
============ CANDIDATE FILTERING =============
candidate_workspace: -1.00 1.00 -1.00 1.00 -1.00 1.00 
min_aperture: 0.0000
max_aperture: 0.0850
==============================================
============ CLUSTERING ======================
min_inliers: 0
==============================================

Processing cloud with 76290 points.
Voxelized cloud: 0
Calculating surface normals ...
num_threads: 4
[pcl::NormalEstimationOMP::compute] input_ is empty!
 runtime(computeNormals): 0.0000
camera: 0, #indices: 0, #normals: 0 
Calculated 0 surface normals in 0.0000s (mode: OpenMP).
Reversing direction of normals that do not point to at least one camera ...
 reversed 0 normals
 runtime (reverse normals): 3.123e-06
Plotting normals for different camera sources

wonwon0, 2022-11-24:
Hi, I have the exact same issue. Ill post here if i find what's  going on. Update me if you find it too please!

I do not have this issue with the tutorials, only when using either the tutorial example with my own .pcd files or when using gpd_ros while subscribed to a pointcloud2 message.

xiaofeiso, 2022-11-24:
OK, if I find a solution, I will reply it.

wonwon0, 2022-11-24:
Ho wow, look at that!

 workspace: -1.00 1.00 -1.00 1.00 -1.00 1.00
 candidate_workspace: -1.00 1.00 -1.00 1.00 -1.00 1.00

My cloud happened to fall outside of this region.
Now it's not generating the same output. Normals are able to be compputed but i dont have a result yet (maybe i bugged something else while fideling into stuff...

wonwon0, 2022-11-24:
i confirm it fixed the issue for me. 

Be aware though, if you run the example, the window is made to visualize an object located at 0 0 0. It's normal that you have to change your point of view in the 3D viewer in order to see what's going on.

xiaofeiso, 2022-11-25:
Did you modify it:eigen_params.cfg，-> workspace = -1.0 1.0 -1.0 1.0 -1.0 1.0,  i don't know how to make it success.

xiaofeiso, 2022-11-25:
How do I know to modify the appropriate value? pcl_viewer has no way of knowing the rectangular extent of the point cloud.

xiaofeiso, 2022-11-25:
ok, the error is solved,but nomal ,candidate grasp not generated, Do you mean to modify camera_position？

closed 2022-12-02 by xiaofeiso