atenpas/gpd #141: double free or corruption (out)

shen-shui, 2024-04-24:
When I use gpd to grab the first item, an error will be reported.
```
robot@WP:~$ roslaunch robot_sim gpd_run.launch type:=2 topic:=/cloud_sample
... logging to /home/robot/.ros/log/4f386c5a-0183-11ef-b6c9-3171f884d3a1/roslaunch-WP-19219.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://WP:43347/

SUMMARY
========

PARAMETERS
 * /detect_grasps/cloud_topic: /cloud_sample
 * /detect_grasps/cloud_type: 2
 * /detect_grasps/config_file: /home/robot/packa...
 * /detect_grasps/rviz_topic: plot_grasps
 * /detect_grasps/samples_topic: 
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /
    detect_grasps (gpd_ros/detect_grasps)

ROS_MASTER_URI=http://localhost:11311

process[detect_grasps-1]: started with pid [19234]
Init ....
============ HAND GEOMETRY ======================
finger_width: 0.01
hand_outer_diameter: 0.12
hand_depth: 0.06
hand_height: 0.02
init_bite: 0.01
=================================================
============ PLOTTING ========================
plot_normals: true
plot_samples false
plot_candidates: true
plot_filtered_candidates: false
plot_valid_grasps: false
plot_clustered_grasps: false
plot_selected_grasps: true
==============================================
============ CLOUD PREPROCESSING =============
voxelize: true
voxel_size: 0.003
remove_outliers: false
workspace: -1.00 1.00 -1.00 1.00 -1.00 1.00 
sample_above_plane: false
normals_radius: 0.030
refine_normals_k: 0
==============================================
============ CANDIDATE GENERATION ============
num_samples: 30
num_threads: 4
nn_radius: 0.01
hand axes: 2 
num_orientations: 8
num_finger_placements: 10
deepen_hand: true
friction_coeff: 20.00
min_viable: 6
==============================================
============ GRASP IMAGE GEOMETRY ===============
volume width: 0.1
volume depth: 0.06
volume height: 0.02
[... 204 lines cut ...]
==============================================
============ CANDIDATE FILTERING =============
candidate_workspace: -1.00 1.00 -1.00 1.00 -1.00 1.00 
min_aperture: 0.0000
max_aperture: 0.0850
==============================================
============ CLUSTERING ======================
min_inliers: 0
==============================================

Created GPD ....
[ INFO] [1713885332.349522727]: Waiting for point cloud to arrive ...
view_points:
0
0
0
[detect_grasps-1] process has died [pid 20941, exit code -11, cmd /home/robot/catkin_ws/devel/lib/gpd_ros/detect_grasps __name:=detect_grasps __log:=/home/robot/.ros/log/4bb36a0c-0184-11ef-b6c9-3171f884d3a1/detect_grasps-1.log].
log file: /home/robot/.ros/log/4bb36a0c-0184-11ef-b6c9-3171f884d3a1/detect_grasps-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```
Can you help me analyze what the problem is?

artyom-boyarov, 2024-06-18:
@shen-shui: I encountered the exact same issue as you. I integrated GPD into a ros 2 service and was using GPD as an external library. I removed the flags ' -march=native -msse4.2 -mavx2 -mfma -flto' from line 29 of GPD's CMakeLists.txt, and the issue was resolved. The last post in this issue: https://github.com/atenpas/gpd/issues/88 helped me.

torydebra, 2024-08-08:
Same issue here, but it happens to be system-dependent: on a pc removing those flags solves the problem, while on another pc the problem persists. The input I give is the same on the two machines.
Any more clues about this issue?