atenpas/gpd #142: Segmentation fault (on Ubuntu 22.04)

Ca11m3X, 2024-09-30:
Hi, I'm trying to install GPD on Ubuntu 22.04 but I have a **"Segmentation Fault"** issue when running the example `./detect_grasps ../cfg/eigen_params.cfg ../tutorials/krylon.pcd`. 

[image]

I just install by default. There are some information about my hardware:

[image]

Thanks for your support.

atenpas, 2024-09-30:
Can you comment out [this line](https://github.com/atenpas/gpd/blob/master/CMakeLists.txt#L29) and try again?

Ca11m3X, 2024-10-01:
> Can you comment out [this line](https://github.com/atenpas/gpd/blob/master/CMakeLists.txt#L29) and try again?

Thanks for your reply, but when I comment whole the line, it caused some errors: `‘omp_get_wtime’ was not declared in this scope` and `recompile with -fPIC`. Then I fix this by `set(CMAKE_CXX_FLAGS "-fopenmp -fPIC")` only, it built successfully but it still `Segmentation fault`.

atenpas, 2024-10-01:
Could you try https://github.com/atenpas/gpd/issues/103#issuecomment-734152660?

Ca11m3X, 2024-10-01:
> Could you try [#103 (comment)](https://github.com/atenpas/gpd/issues/103#issuecomment-734152660)?

I had tried this solution before but not work.

atenpas, 2024-10-02:
Could you run it in gdb and post the backtrace?

Ca11m3X, 2024-10-03:
> Could you run it in gdb and post the backtrace?

Sorry for slow response, I had ran the `./detect_grasps ../cfg/eigen_params.cfg ../tutorials/krylon.pcd` in gpd with build setup `set(CMAKE_CXX_FLAGS "-fopenmp -fPIC -Wno-deprecated -Wenum-compare -Wno-ignored-attributes -std=c++14")` then the error caused by the line as below:
[image]
If its not enough for finding the issue, then let me know.
Thanks for ur support.

atenpas, 2024-10-03:
This seems to be related to PCL. You can have a look at https://github.com/PointCloudLibrary/pcl/issues/5237.

Ca11m3X, 2024-10-04:
> This seems to be related to PCL. You can have a look at [PointCloudLibrary/pcl#5237](https://github.com/PointCloudLibrary/pcl/issues/5237).

Thanks for ur reply, after I upgrade to PCL 1.13.1 instead PCL 1.12.0 on Ubuntu 22.04, it works completely. Additionally, the issue is not come from `CMAKE_CXX_FLAGS` so I set it as default `set(CMAKE_CXX_FLAGS "-O3 -march=native -mtune=intel -msse4.2 -mavx2 -mfma -flto -fopenmp -fPIC -Wno-deprecated -Wenum-compare -Wno-ignored-attributes -std=c++17")`.

I really appreciate your supports.

closed 2024-10-11 by Ca11m3X